rbroker/ecodan-ha-local #1: [Support] I can´t connect with my heat pump

limkinZero, 2023-08-01:
Hi @rbroker,

I'm trying to use your code to connect to my Ecodan FCT6 but I can't see data on my home assistant.

My esp32 card is a WROOM Devkit and in principle the TX (GPIOP1) and RX ( (GPIOP3) pins for serial communication I think they should be fine but I don't receive any data.

Could you help me with this? I'll give you a couple of screenshots and the diagnostic log.

**Device info**
[image]

**Overview**
[image]

**Logs**
```logs
[15:49:37] Configuration parameters loaded from NVS
[15:49:37] Initializing WiFi connection...
[15:49:37] Waiting 500ms for WiFi connection...
[15:49:38] WiFi connection established!
[15:49:38] Updated UTC time from NTP
[15:49:38] Regular startup mode, initializing web-server...
[15:49:38] Initializing HeatPump with serial rx: 3, tx: 1
[15:49:38] Initializing MQTT...
[15:49:38] MQTT user 'admin' has configured password, connecting with credentials...
[15:49:38] Successfully established MQTT client connection!
[15:48:43] Published homeassistant auto-discovery topics
[15:48:43] Ecodan HomeAssistant Bridge startup successful, starting request processing.
[15:48:43] Login cookie is unset, redirecting to login page
[15:48:52] Successful device login, authorising client.
```

[ESP32 WROOM Pinout](https://m.media-amazon.com/images/I/61y7J88hMkL._SL1067_.jpg)

rbroker, 2023-08-01:
Hi,

I'm not 100% sure, but it may be because I've set things up to use UART1, and it looks like you've wired things up to use UART0 on the WROOM.

It may work if you change the code in "ehal_hp.cpp":

Line 15:
```
HardwareSerial port = Serial;
``` 
Should tell things to use UART0, I think? But I've only ever tried it using UART1 on my side.

limkinZero, 2023-08-02:
Hi,

I´ve tried everything. Change pins and use UART2 with RX2 (16) and TX2 (17) but it remains the same. Does not receive anything from heat dump.

```cpp
HardwareSerial port = Serial2;
```

limkinZero, 2023-10-22:
I've buy the same board like you (lolin S2 mini). When it arrives, I will check.

rbroker, 2023-10-30:
Just FYI, I pushed some changes to try to improve compatibility with the WROOM boards, I don't know if it will help with your issue, but it moves the serial tx/rx pins to ones which should work for the WROOM and fixes an issue that I wasn't explicitly setting the TX pin as an output.

limkinZero, 2023-10-30:
Hi Richard,

I just tried again with the last version (v0.0.7) with default settings (TX to GPIO26 ant RX to GPIO27) and the result is the same. I've issues with the reception. 
[image]

I don't know if I'm missing something or if the pins are okay. I attach a photo. Is there any way to debug if there is an error with the UART serial port and the RX and TX pins? I think the LOLIN plate will arrive this week.

rbroker, 2023-10-30:
> I don't know if I'm missing something or if the pins are okay. I attach a photo. Is there any way to debug if there is an error with the UART serial port and the RX and TX pins?

Hmm, so from that picture you're still just not receiving any replies from the heatpump. Debugging is a bit awkward, because I don't think there's much we can easily do from software, from the point of view of the code it's successfully sending data on the GPIO pins, but it's just not getting a response.

One option might be to wire up and LED on the between TX pin on the ESP32 side, and RX on the heat pump, and between RX on the ESP32 and TX on the heat pump. At 2400 baud it should be slow enough that you can see it flash when there's data being transmitted. That might let you confirm if the ESP32 just isn't able to send any data, or if it's really that responses signals from the heat pump aren't being received. (If you have one, a logic analyser might be a more high-tech solution to that).

limkinZero, 2023-10-30:
Hi Richard,

I tested again with other esp32 (Chinese clon of spressif wroom 32) and the
first time I had the same error. Then, I switch in my board the TX and RX
pins and Eureka! the board starting received data.

So, I think it was my mistake. In my HA integration, I've some entities
Whose values ​​I don't understand. Especially the temperature of zone 1. If
you like, you can close this issue and we will discuss the other topics in
another thread.

Regards.

rbroker, 2023-10-31:
> I've some entities Whose values ​​I don't understand. Especially the temperature of zone 1. If you like, you can close this issue and we will discuss the other topics in another thread.

Sure, no problem, I'm glad you found the issue in the end!

My system only has 1 zone, so perhaps if you have 2 zones some modification is needed but it's hard for me to say. I'm happy to try and help in another thread if you'd like (my names for some entities maybe are also not very good / accurate, so maybe it is not helping?)

closed 2023-10-31 by rbroker