rbroker/ecodan-ha-local #3: Connection

Sewek0, 2023-10-28:
After setup configuration ESP reboots and apears in network but there is no way to connect to ESP32 also mqtt do not work.
I tryed with Heat pump connect and disconect too.
Device: ESP-32S WROOM-32

rbroker, 2023-10-28:
Hi, 

I've purchased a WROOM dev board to give it a try as there was another issue reported where it didn't seem to work with that hardware. In the meantime if you discover some issue / incompatibility in the code, I'm happy to accept a pull request!

When you say there is no way to connect to it, does that mean when you try to get back to the configuration page in your browser over your normal WiFi the connection just seems to hang? Or does it immediately report some connection failure message?

rbroker, 2023-10-29:
If I open up the serial monitor in the Arduino IDE, I see it boot-looping and printing messages like this:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
E (635) uart: uart_set_pin(692): tx_io_num error
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4008d7ac
file: "<esp root>\esp32\hardware\esp32\2.0.14\cores\esp32\esp32-hal-uart.c" line 214
func: uartBegin
expression: uart_set_pin(uart_nr, txPin, rxPin, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE)

abort() was called at PC 0x4008d7af on core 1
```

Do you see a similar output?

It seems like the default GPIO pin settings cause a fatal error, because on the WROOM boards pin #34 (the default for the TX setting) is input-only. If I change this to use RX Pin: 32, TX Pin: 33 it avoids the issue.

The default pin settings are in `ehal_config.cpp`:
```
        config.SerialRxPort = prefs.getUShort("serial_rx", 32U);
        config.SerialTxPort = prefs.getUShort("serial_tx", 33U);
```

If you use "Tools" > "Erase all flash before sketch upload: Enabled", and re-flash the board with those changes it should hopefully clear out any bad pin settings (but you'll also have to reconfigure the WiFi settings unless you also hard-code them into the `ehal_config.cpp` file.

Sewek0, 2023-10-30:
Thanks for response aI already buy S2 min I will give it a try.
I also wander if it will be a way to use  WT32-ETH01 or Olimex ESP32??

rbroker, 2023-10-30:
> I also wander if it will be a way to use  WT32-ETH01 or Olimex ESP32??

Hmm, sorry I'm not sure, as I don't have either of these boards. WT32-ETH01 does seem to compile, and so does the olimex devkit board config so it may work.

I've just pushed some updated code which I think should now use some more compatible pins (RX27, TX 26) for the ESP32 WROOM, and fixes an issue which I think could prevent the serial port working correctly.

Sewek0, 2023-11-01:
Hi I have check the communication on three devices and I alvays got same error.
My ESP32 DEVKIT1 is visible at router but there is no way to connect to it. 

this is reboot in loop
[image]

limkinZero, 2023-11-02:
I think is a pin configuration problem. The error shows that you are using
gpio0, a pin with disallowed output.

You should use other pins like gpio26 and gpio27 for tTX and RX.

Sewek0, 2023-11-02:
Ok, I made some investigation yesterday adn I solved the problems. I was my mistake. Now I can connect to ESP over local network and set up everyting. I can not connect to Heat pump. 
I use WT32-ETH01. For connection with heat pump I choose pin 17 and 5 it is good for debuging because producer attached the led to this pins so I can see is ther any signals from heat pump.
There are nothing ther. So meybe ther is an option in heat pump to enable this communication. 
Before ESP I tryed to use Melcloud but with no success at all. My heat Pump is quite olb. It is FTC 4.

rbroker, 2023-11-02:
> There are nothing ther. So meybe ther is an option in heat pump to enable this communication.
> Before ESP I tryed to use Melcloud but with no success at all. My heat Pump is quite olb. It is FTC 4.

Hmm, yes if you're sure it's all wired up correctly but there is still no response from the heat pump to ESP32, you can try enabling the `Dump Serial Packets:` option on the configuration page then saving the settings. 

It should cause any connection packets sent by ESP32 to be displayed on the "Diagnostics" page. You can copy-paste the output to this ticket, and maybe there is some simple change which could be made to support the FTC4, but I can't make any promises!

Sewek0, 2023-11-02:
Meydy ther is any hidden menu in Heat pump to enable this connection?

rbroker, 2023-11-02:
> Meydy ther is any hidden menu in Heat pump to enable this connection?

It could be this, but I don't know about the FTC4 but it would be strange. The best thing would be to find the FTC4 installation manual and check it for more details?

I think most likely is either that there is some difference about how the connection request must be sent for older FTC, or some issue with the serial RX/TX wires. I'm sorry I don't think there's any other advice I can give as my system is a newer version.

Sewek0, 2023-11-02:
I will continue investigation and I will share results here so please do not close this thread.
Also it will be very useful if it will be possible to use Ethernet connection instead wifi but I do not know what I need to thange in your code to get it. I mostly use ESPHome for my project.

drnichols, 2023-12-03:
@rbroker I have been testing on a WROOM32 and had some very strange intermittent failures; initially I believed it was an issue connecting to wifi (serial was logging "wifi:association refused temporarily comeback time 0 msec" repeatedly), after many hours chasing my tail I noticed that the WROOM32 was more stable connected via USB to my laptop and ultimately when the virtual serial connector was active.

Today I found this thread https://forum.arduino.cc/t/esp32-wifi-wont-connect-without-the-serial-being-open/1065685/3 which identified an issue specific to the WROOM32 and that the enable pin needed to be pulled up to 3.3V. Since pulling the enable pin up with a 4.7k resistor connected to 3.3V it has been solidly connected to my wifi for the past 8hrs+.

I may not be related to your issue @Sewek0 but I thought worth logging here as it was driving me mad for the past week or so!

I am using the updated version of your code and using pins, RX26, TX27 which are working fine with my WROOM32. 

Thank you @rbroker for sharing this project I've been looking for a local control solution for my ecodan's for some time and your project is the most polished and usable I've found. You even wrote comprehensive documentation! Great work.

rosstechinc12, 2025-01-09:
i also have this issue can connect to its own network but it will not connect to my network. it hangs on “loading ssids” and will remain doing so until reboot. it worked for me once suddenly disconnected and then has never been able to reconnect even after reflashing. (esp23-wroom32 az-delivery) i tried the resistor fix as mentioned above to no avail.

rosstechinc12, 2025-01-09:
E (26878) wifi:sta is connecting, return error
E (26882) wifi:sta is connecting, return error
E (26886) wifi:sta is connecting, return error
E (26890) wifi:sta is connecting, return error
E (26895) wifi:sta is connecting, return error
E (26899) wifi:sta is connecting, return error
is being spammed to the console

Babka1111, 2025-03-15:
Hi, can you show the ecodan-ha-local.ino settings for WT32-ETH01. Thank you

Rudo1111, 2025-03-16:
> Ok, I made some investigation yesterday adn I solved the problems. I was my mistake. Now I can connect to ESP over local network and set up everyting. I can not connect to Heat pump. I use [...]
Hi, can you show the ecodan-ha-local.ino settings for WT32-ETH01. Alternatively the basic ecodan-ha-local.ino.merged.bin file for WT32-ETH01. thanks