rbroker/ecodan-ha-local #4: [Request] Set DHW Temperature

DD-bep, 2023-10-31:
Hi, it's a really nice project :)
Would it be possible to add dhw temperature control from home assistant?
So I could increase the dhw storage temperature when the photovoltaic has excess production.
I tried to look at the code of BartGijsbers project but I couldn't add this function to your code, I'm not a good programmer :/
Thank you

limkinZero, 2023-10-31:
Hi, I think this feature is the only thing that would be missing to match the official integration of melcloud in home assistant.

I have also been reviewing the code and I think it should be possible. if I can find time and try it.

rbroker, 2023-10-31:
It seems pretty quick to add it. I've made a quick change which seems to work, but I'll do a little bit more testing with it first.

rbroker, 2023-11-01:
I just pushed commit dcb991c which should add a new control for this:

[image]

It allows the range 40-60 degrees celsius, which is what the FTC6 installation manual claims is the allowed range, there is also a note which says that if the target temperature is set over 55 degrees, the DHW must not be restarted until the temperature drops below 50 degrees to protect the device  (page 36) - I haven't added any code to force this, and I don't know if the FTC6 will do it by itself, so be a bit careful with any automations using it!

Note, there is another commit before that which makes some changes to the UART handling to use a bit less power, so sorry if I broke something and you have any problems with that after updating.

closed 2023-11-01 by rbroker

limkinZero, 2023-11-01:
Hi, Richart. Awesome work. I´ve tested the new version and it seems to work well.

The only thing I think is that you should send MQTT info as **water_heater.** If you send the information according to the API, the control that can be used in HA is much better, since the services in HA could be called to heat water through automations. Additionally, you could put the modes (boost, eco, ...) in the graphic component. If you need help with that, tell me.

You can see more info [here](https://developers.home-assistant.io/docs/core/entity/water-heater/)

In melcloud native is like this.

rbroker, 2023-11-01:
> The only thing I think is that you should send MQTT info as water_heater.

Oh yes, that's a much better idea, thank you! It should be implemented by 84d4dff. (Sorry, you may have to manually publish a MQTT config topic with an empty payload to get rid of the previous "number" control).