rbroker/ecodan-ha-local #5: [Question] Flow rate

limkinZero, 2023-11-02:
Hi Richard,

First of all, thank you. You are doing very good work.  I've checked the source code of file `ehal_hp` and i see that you collect flow rate info. What does this info mean? This param show the flow rate in litres/minute of water pump?

```c++
case GetType::FLOW_RATE:
                status.FlowRate = res[12];
                break;
```

Then, i haven't seen that you publish that info to mqtt. This is so?

rbroker, 2023-11-02:
> This param show the flow rate in litres/minute of water pump?

Yes, I believe the value is in l/min. 

> Then, i haven't seen that you publish that info to mqtt. This is so?

Yep, it would need to be added to `publish_homeassistant_auto_discover()` and `publish_entity_state_updates()` in `ehal_mqtt.cpp`. I think originally I planned to add it to the data displayed on the status web pages of the ESP32, but early in development I had a lot of memory shortage problems serving the HTTP content so those pages only display a small amount.

rbroker, 2023-11-03:
Note: Was added in PR #6, so I'll close this off.

closed 2023-11-03 by rbroker