rbroker/ecodan-ha-local #9: Make heating mode configurable

Algram, 2023-11-09:
Hi there,

I've been using your lib for quite some time now. I have a humble request: Is there any way to make the "Heating Mode" configurable? So that I can remotely switch between "Compensation Curve" and "Room Temperature Controlled"?

rbroker, 2023-11-09:
Hi,

> Is there any way to make the "Heating Mode" configurable? So that I can remotely switch between "Compensation Curve" and "Room Temperature Controlled"?

Unfortunately, I don't currently know the required flags for changing the heating mode (flow temp, room temp, compensation curve).  So unless someone is able to figure out the correct flags for the "set" command I wont be able to support it.

Algram, 2023-11-09:
@rbroker I've opened a Draft PR https://github.com/rbroker/ecodan-ha-local/pull/10.

In my mind it should work, since in the protocol decode it is specified at comand `0x32 HC1 or HC2`. Did you test that already as well? It does not seem to work with my PR, but that could very well be my lack of C++ knowledge.

limkinZero, 2023-11-09:
Hi @rbroker , 

Have you checked the [BartGijsbers repository.](https://github.com/BartGijsbers/CN105Gateway/tree/main#implemented-settings) It says that it has implemented that function.

[image]

rbroker, 2023-11-09:
Thanks @Algram & @limkinZero, I think that was enough puzzle pieces for me to get it working on my end. I pushed some changes in 1e42acc which seems to work for toggling at least between the 3 heating modes (I'm a bit reluctant to support the extra modes from the other repo as my unit can't do cooling, and floor dryup is only really useful for immediately after installing underfloor heating (if I understand correctly)

[image]

closed 2023-11-09 by rbroker