rbroker/ecodan-ha-local #18: Error esp32 don't reconnect to Wifi AP

limkinZero, 2024-01-05:
Hi @rbroker ,

Merry Christmas! I think there is an issue when reconnecting to the Wi-Fi network if the router reboots.

This Christmas I noticed that when I was configuring my new router, the Ecodan local integration was not connecting to my Wi-Fi network. I had to press the reset button on the board for it to come back.

mharizanov, 2024-01-06:
I had the same issue with my home setup of 4 ASUS routers in a mesh setup. Fixed it with a more aggressive reconnection strategy, see my branch here: https://github.com/rbroker/ecodan-ha-local/compare/main...mharizanov:ecodan-ha-local:main 
The code is too quick and dirty to request a merge though.

rbroker, 2024-01-06:
I'll take a look. The code which resets the WiFi configuration tries to do that in case people's WiFi setup changes, so they don't need to physically access / re-flash things. I guess the restart from your fork is needed for that @mharizanov, which is missing from mine.

Out of interest was rebooting every 24h needed for you? Or just being defensive?

rbroker, 2024-01-06:
> I understand, but that may also pose a security risk.

Yeah, fair enough. It should preserve the password for that reason, but I can see that no everyone would want that behaviour.

> The 24-hour restart timer is just a precaution

Understood, thanks!

rbroker, 2024-01-06:
I've made some changes in commit [#97243e1](https://github.com/rbroker/ecodan-ha-local/commit/97243e12739de4324fe3409bb354e640c26c93b9) which extends the time we wait for WiFi reconnects to 10 mins, and will reset the board after those 10 mins if the WiFi connection hasn't been re-established (in theory it should figure it out if the SSID/password haven't changed because of [this](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/wifi.html#setautoreconnect)).

It'll now log when the WiFi connection is lost, and how long it takes to recover it too:
```
[00:07:54] WiFi disconnection detected... allowing up to 10 minutes to recover...
[00:08:42] WiFi connection re-established after (48) seconds.
```

There's now also a setting to stop the ESP32 falling back to captive portal mode if the connection can't be established on boot ("Auto-Reset WiFi Settings") for people who don't want it that (it'll just reboot every 10 minutes until the originally-configured WiFi network comes back)

Hopefully that improves things!

closed 2024-01-06 by rbroker

limkinZero, 2024-01-06:
Thanks for your work Robert. I'll test it

limkinZero, 2024-01-10:
> This version works well in my Asus router mesh WiFi setup. Thanks!

Yes, me too! Works with my Asus wifi mesh.