rbroker/ecodan-ha-local #20: Problem compiling

Sipowitz, 2024-02-20:
Hi, 

I'm new to using the Arduino IDE and am obviously doing something stupid as I am getting an error while trying to compile.

I have downloaded, unzipped and opened the .ino file. I have installed the three listed dependencies but I am getting a file not found error for these libraries

`#include <chrono>`
`#include <deque>`
`#include <memory>`
`#include <mutex>`
`#include <thread>`

Obviously I need to add them but I am not sure where to download them from?

Many Thanks

rbroker, 2024-02-20:
Hi @Sipowitz,

These header files are part of the C++ language, it will depend a bit which board you are using, but if it's an ESP32 based device they should be available after:
- Installing the Arduino packages for ESP32 as listed [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html)
-  Selecting to build for your ESP32 board.

It's possible that some boards do not include support for C++, in which case you might be out of luck!

Thanks,
Richard

rbroker, 2024-03-09:
Closing issue to clean up my dashboard a bit :)

closed 2024-03-09 by rbroker