rbroker/ecodan-ha-local #21: Adding Z2 Flow Temperature control

lunarok, 2024-03-01:
Hi,

In my case, I control the Zone 2 by flow temperature.
I see that the actual temperature is here, this command (set flow temperature) exist for Z1 and I will like to try to add it.
I find the PR with Z1 flow control addition, and will like to know if a "simple copy paste" of what is in the PR by replace Z1 by Z2 will work ?

https://github.com/rbroker/ecodan-ha-local/pull/16

lunarok, 2024-03-01:
Got it working, I have created PR https://github.com/rbroker/ecodan-ha-local/pull/23

lunarok, 2024-03-02:
Something is missing. There is no error visible in logs, command is created can be set but at next refresh back to previous value. So I think command as added doesn't work.
Pretty sure it's because of cmd[6] = static_cast<uint8_t>(SetHpMode::FLOW_CONTROL_MODE); on line 369. This value doesn't exist for Z2 actually (but it is on the Ecodan)
I'm trying to understand how to add it for Z2 the information command and also the select command.

rbroker, 2024-03-05:
Hi!

Thanks for the PR and sorry for the delayed reply. I only have a 1 zone system so I can't test any change to the second zone.

Is the state of the Z2 set point correct when you change it using the manufacturer controls? It's definitely possible the offsets for the set commands are not defined correctly, there seemed to be some uncertainty in other projects about the correct bits to use for Z2.

lunarok, 2024-03-05:
Hi,

So, fex confirmations/informations :
- z2_flow_temp_target : yes that information from ehal is correct
- z2_room_temp : this one for information look wrong, I have 28,25° displayed, while it's only 20
- sh_mode : it's different in Z1 and Z2, but I found only one here

And a guess, it's that it can be the issue with the command I add. Because it retrieve the mode of operation from z1 that in my case is "Heat Target Temperature" while sending a command for setting the flow temperature. It's not coherent.
Not sure about if it's real this, but sure thing is that Z1 and Z2 can have different operation mode, in my case Z2 doesn't have the Heat Target Temperature.
So I think a start will be to find the information for operation mode of Z2

rbroker, 2024-03-09:
@lunarok, if you can enable "Dump Serial Packets" for your system, and share the output + current zone 2 room temperature when the next state update happens, we can check the 0xB result message for anything which looks like it might be the correct zone 2 temperature.

lunarok, 2024-03-09:
[19:10:31] RES { .Hdr { fc, 62, 2, 7a, 10 } .Payload { b, 7, d0, 7, d0, f0, c4, b, 9, c4, 7e, 5e, 0, 0, 0, 0 } .Chk { f1 } }
b here it's the 0x0B you are looking for right ?

At time of this info, I can see on sensor display :
Z1 : 20°
Z2 : 20°
(both same)

Do I understand correctly that this :
status.Zone1RoomTemperature = res.get_float16(1);
status.Zone2RoomTemperature = res.get_float16(7);

Will have to be more 
status.Zone1RoomTemperature = res.get_float16(1);
status.Zone2RoomTemperature = res.get_float16(3);

Not sure of the index, but can see 7, d0, 7, d0, like twice the same info

lunarok, 2024-03-09:
I'm digging your code, start to understand more I think. So for Z2 temp, I think best is to let the other issue discuss it.
Here to get the Z2 flow temp target (or room temp, I think it same), I'm looking first to get the "set_heating_cooling_mode Z2"

If I'm right, it's 0x26 frame that is providing mode
[19:28:31] RES { .Hdr { fc, 62, 2, 7a, 10 } .Payload { 26, 0, 0, 1, 0, 1, 0, 1, 14, b4, 7, 6c, 11, 94, 0, 0 } .Chk { 9 } }

I think it's the number 7th of payload the Z2 mode.
Because it's 1 and my Z2 is in flow temp target mode (cannot be in room temp)
https://github.com/rbroker/ecodan-ha-local/blob/8b523ed058378c7262d7f9352a47047baf2f0e1c/ecodan-ha-local/ehal_hp.h#L59

So could get a new info with the mode for Z2. Once that one exists, I think that modifying the command of this PR with the value of this Z2 mode, must work

rbroker, 2024-03-09:
> I think it's the number 7th of payload the Z2 mode.
> Because it's 1 and my Z2 is in flow temp target mode (cannot be in room temp)

Interesting does that mean your Zone1 is set to target room temperature mode, instead of target flow temperature? Or does the Zone2 setting always override both zones?

Actually, this might be in the FTC6 installation manual, I'll have to read it again.

lunarok, 2024-03-10:
Both zonez have different mode selection.
For Zone 1, I can have (from the main unit panel: room temp, flow temp, compensation curve, cool flow temp (cannot set cool room temp)
And Zone 2, I have flow temp or compensation curve. No heat room temp or cold flow/room temp

That's why I "think" the command I add doesn't work.
https://github.com/rbroker/ecodan-ha-local/pull/23/files#diff-355f0d541b03ab72636075f8a8df3e1671df61eb042abb6b843391d305bb53dfR369
The command send used the mode but for Z1 so room temp, but it's not allowed for Z2 on my unit at least.
It's why I think first step is to add mode for Z2 as for Z1. Then the command to set control will use this new parameter that is valid

Am I clear ? Just to be sure, because myself I'm "guessing" a lot here :)

rbroker, 2024-03-11:
Hmm ok, so in that case the following should work (I believe it's possible to e.g. configure the desired target flow temperature even if the system is not currently using target flow temperature mode for that zone).

So, the 4 commands would be something like:
- Set Zone 1 Room Temperature
```
        Message cmd{MsgType::SET_CMD, SetType::BASIC_SETTINGS};
        cmd[1] = SET_SETTINGS_FLAG_ZONE_TEMPERATURE;
        cmd[2] = static_cast<uint8_t>(SetZone::ZONE_1);
        cmd.set_float16(newTemp, 10);
```
- Set Zone 1 Flow Temperature
```
        Message cmd{MsgType::SET_CMD, SetType::BASIC_SETTINGS};
        cmd[1] = SET_SETTINGS_FLAG_ZONE_TEMPERATURE;
        cmd[2] = static_cast<uint8_t>(SetZone::ZONE_1);
        cmd[6] = static_cast<uint8_t>(SetHpMode::FLOW_CONTROL_MODE);
        cmd.set_float16(newTemp, 10);
```
- Set Zone 2 Room Temperature
```
        Message cmd{MsgType::SET_CMD, SetType::BASIC_SETTINGS};
        cmd[1] = SET_SETTINGS_FLAG_ZONE_TEMPERATURE;
        cmd[2] = static_cast<uint8_t>(SetZone::ZONE_2);
        cmd.set_float16(newTemp, 10);
```
- Set Zone 2 Flow Temperature
```
        Message cmd{MsgType::SET_CMD, SetType::BASIC_SETTINGS};
        cmd[1] = SET_SETTINGS_FLAG_ZONE_TEMPERATURE;
        cmd[2] = static_cast<uint8_t>(SetZone::ZONE_2);
        cmd[6] = static_cast<uint8_t>(SetHpMode::FLOW_CONTROL_MODE);
        cmd.set_float16(newTemp, 10);
```
If that doesn't work I guess there would be some detail missing about how the zone 2 configuration is set.

thoniodu10, 2024-03-26:
Hi guys,

Just to know, it's still not possible to set the Z2 temperature target, right ?
so, still no climate entity as for Z1 ?

Anyway, it's great to have the Z2 current temperature !

Thanks,
Kind Regards

merlin051, 2024-04-09:
In a Z1 & Z2 setup I have both configured with the same flow temp, but the point at which it calls for heat is different.

Downstairs UFH (Zone1) - multi room stats can trigger the zone valve to the UFH manifold
Upstairs Radiators (Zone 2) - Digital Mitsubishi Wireless Thermostat (Target Temp can be set on the controller and set on te MelCloud, controller also shows the current the temperature and has hot water boost button)