rbroker/ecodan-ha-local #22: Adding Energy metrics for Cooling mode (Consummed/Produced)

lunarok, 2024-03-01:
Hi,

I know that the heatpump produce also value for Cooling Mode like for Heating and DH on Energy Consummed and Produced.
Any guidance possible of how to add it ? EnergyConsumedHeating / EnergyDeliveredHeating -> EnergyConsumedCooling / EnergyDeliveredCooling

rbroker, 2024-03-05:
Hi @lunarok,

I don't have cooling support on my system. It seems quite possible these values would be included in the data for commands 0xA1 and 0xA2. If you can enable the debug logs and wait for an update, and confirm what the FTC control panels says is the energy consumed/delivered for cooling in the last 24h, maybe something will stand out?

Thanks,
Richard

lunarok, 2024-03-05:
Hi,

I can see that :
[13:33:12] CMD { .Hdr { fc, 42, 2, 7a, 10 } .Payload { a1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 } .Chk { 91 } }
[13:33:12] RES { .Hdr { fc, 62, 2, 7a, 10 } .Payload { a1, 18, 3, 4, 0, b, 59, 0, 0, 0, 0, 5, 6, 0, 0, 0 } .Chk { e3 } }
[13:33:12] CMD { .Hdr { fc, 42, 2, 7a, 10 } .Payload { a2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 } .Chk { 90 } }
[13:33:12] RES { .Hdr { fc, 62, 2, 7a, 10 } .Payload { a2, 18, 3, 4, 0, 23, 5f, 0, 0, 0, 0, e, 1d, 0, 0, 0 } .Chk { a4 } }

Based on https://github.com/rbroker/ecodan-ha-local/blob/97243e12739de4324fe3409bb354e640c26c93b9/ecodan-ha-local/ehal_hp.cpp#L542, I understand that the full line a1 is energy consummed, with part of the message being the heat, part the dhw. So a part could be the cold.
Issue that as it's winter, I can tell that in both line the total for cold is 0 :) But seems in payload there is 2, one "between" Heating and DHW, one after DHW. Is my understanding right ? Not very fluent in hexadecimal manipulation.
I can try to set the cold mode if required to see which subset of 0 move, but need to pick the right day as the weather is pretty chaotic

rbroker, 2024-03-09:
The cooling stats should be available from [8b523ed](https://github.com/rbroker/ecodan-ha-local/commit/8b523ed058378c7262d7f9352a47047baf2f0e1c) onwards, let me know if you have any issues with it.

Should be available over MQTT as "cooling_consumed" and "cooling_delivered".

closed 2024-03-09 by rbroker

lunarok, 2024-03-09:
Yes, have updated the firmware and can see the informations showing up in MQTT. Of course these days it's 0 for both, will come back in summer if some issue with values :)
Thanks