rbroker/ecodan-ha-local #24: Any plans to expose sensors which report MQTT and Heat Pump Connection Status?

jamessimpson007, 2024-03-02:
A massive thanks to you Richard Broker for creating this amazing project. Hilarious that one man can create such a polished interface that works beautifully, whereas an enormous multinational corporation fails miserably to provide any sort of decent control.

Do you have any plans to expose sensors which report the MQTT and Heat Pump Connection status in HA - much like that reported on the browser-based interface?

Thanks, James

limkinZero, 2024-03-03:
Hi, @jamessimpson007. This project already have integration with home assistant and mqtt

jamessimpson007, 2024-03-03:
Hi, sorry perhaps I wasn’t clear. Yes I’ve got all the MQTT sensors and controls in HA which all work brilliantly. I was wondering if it was possible to have extra sensors which report 

1) The connection status of the ESP32 module to the MQTT server
2) The connection of the ESP32 module to the heat pump 
3) The strength of the WiFi signal connection of the ESP32 module

Unless I’m being dumb and missing something obvious I can’t find any way of doing this. That would be the icing on the cake!

Thanks

rbroker, 2024-03-05:
Hi,

Thanks for the kind words, to be clear I had a good amount of help from kind people like @limkinZero, so it's not just my work!

I think it should be possible to add some sensors for the state of the ESP32 (hp connection and WiFi strength).

I think MQTT connection seems maybe not as useful, as we can only report it if MQTT is working already and it doesn't really have any partially working states.

I'll take a look when I get some time!

Thanks,
Rixhard

piplica, 2024-03-06:
On my ESP32-DEVKIT1 everythinh look great, only MQTT Connection is not stable.

lunarok, 2024-03-10:
@piplica are you still facing MQTT stability issues ?
I have this also after reboot/flash, for sometime I can see on mosquitto logs it's connecting/disconnect. When in that MQTT state, web interface is mainly not usable.
Just want to see if you also face this before digging or open issue for help

piplica, 2024-03-11:
> @piplica are you still facing MQTT stability issues ? I have this also after reboot/flash, for sometime I can see on mosquitto logs it's connecting/disconnect. When in that MQTT state, web interface is mainly not usable. Just want to see if you also face this before digging or open issue for help

Hi, it was 2 days conected ion my HP, and I was facing the issue the whole time.
Im using ESP32-DEVKIT1, and was on 16 and 17 Pin, now Im gonna test this on 32 nad 33 PIN, but at the end of March.
Or schould I buy LOLIN S2 Mini?

Thanks

rbroker, 2024-03-11:
> Or schould I buy LOLIN S2 Mini?

It's difficult to say, as it's not clear why the connection quality is not good in your case. When you go to the web view for ecodan-ha-local and "Reload WiFi SSIDs" on the configuration page, what sort of signal strength is is reporting for the ESP32? 

What sort of machine are you using for the MQTT server? I had lots of problems with connection dropouts with a Raspberry Pi on WiFi which went away after I used ethernet for the RPi network connection.

piplica, 2024-03-13:
Hi, in most cases WiFi, and Web interface is available, but Not Mqtt.

pcman75, 2024-03-13:
*we can only report it if MQTT is working already*
We can report a lost MQTT connection by using "MQTT Will message"

rbroker, 2024-03-14:
> *we can only report it if MQTT is working already* We can report a lost MQTT connection by using "MQTT Will message"
Ah, that is a good point thanks. I guess it will race between the ESP and the broker detecting the closed connection as we never close it intentionally on the ESP, but maybe it's still useful.

I'll try to add some more detailed error information to the logging as well for MQTT issues so the reason for any disconnection is now obvious.

jamessimpson007, 2024-03-15:
Although my set up is generally working brilliantly, the MQTT sensors do intermittently go unavailable a few times every day. It seems to just last a few minutes and then they come back online again. I’m not sure whether it’s a wifi reception issue or something else. There seem to be plenty of bars next to the WiFi section in the configuration page. It has been causing some issues with some of my HA automations but I have set up some (hopefully) temporary trigger template sensors to ignore the unknown/unavailable states when it happens. 

I’m no MQTT expert, but could the retain flag be used here for the sensors (not controllers) which might help? I think it used to be possible to do within HA but that caused issues so now can only be done by the publisher?

jamessimpson007, 2024-03-16:
Thanks for that. I’ve updated as described and from the configuration pages all seems well (MQTT connected and green) and sensors all work normally although it’s too early to see if unavailable episodes continue. 

However, I’ve been looking in the Mosquitto logs and I’m getting loads of refused / broken pipe messages, despite there being nothing worrying in the logs of the "Ecodan Home Assistant Bridge”. 

‘''
2024-03-16 15:47:43: Connecting bridge heatpump (192.168.0.25:1883)
2024-03-16 15:47:43: Client local.heatpump disconnected: Connection refused.
2024-03-16 15:48:14: Connecting bridge heatpump (192.168.0.25:1883)
2024-03-16 15:48:15: Client local.heatpump disconnected: Connection refused.
‘''

I’ve done a load of troubleshooting including checking user/pass credentials, trying different user name, changing IP address, checking router settings, re-installing Mosquitto, is anyone else experiencing similar issues?

rbroker, 2024-03-16:
> However, I’ve been looking in the Mosquitto logs and I’m getting loads of refused / broken pipe messages, despite there being nothing worrying in the logs of the "Ecodan Home Assistant Bridge”.

Connection refused means that mosquitto is trying to open out a connection to somewhere else (which seems to exist), but that whatever it is isn't listening for MQTT protocol messages. Is 192.168.0.25 the IP address assigned to your ESP32? Do these attempts to connection out from mosquitto line up with some automation tying to change some heatpump settings?

piplica, 2024-03-16:
Im unable to test it until end of March.

jamessimpson007, 2024-03-17:
> Connection refused means that mosquitto is trying to open out a connection to somewhere else (which seems to exist), but that whatever it is isn't listening for MQTT protocol messages. Is [...]

Thanks again Richard. You’re right, I hadn’t twigged that it was the other way around. Problem solved! Appreciated.

piplica, 2024-03-17:
> Thanks again Richard. You’re right, I hadn’t twigged that it was the other way around. Problem solved! Appreciated.

And was was the solution?

jamessimpson007, 2024-03-17:
yup, I had a bridge set up from an old configuration I had forgotten about, called something similar, which was trying to open a connection from Mosquitto. Richard's post jogged my memory as I had assumed that the error messages meant that his python script was trying to connect with Mosquitto, rather than something else trying to do it the other way around!

piplica, 2024-03-17:
Old ecodan configuration?
Pls describe

piplica, 2024-03-17:
Juste testit localy, thats my output.
After some time MQTT stops working.
```
[14:46:11] MQTT user 'home' has configured password, connecting with credentials...
[14:46:12] Successfully established MQTT client connection!
[14:46:14] MQTT publishing failure: 'homeassistant/sensor/hp_return_temp_809c6a3c9c8/config': -9
[14:46:14] MQTT network disconnection detected trying to publish: 'homeassistant/sensor/hp_return_temp_809c6a3c9c8/config' attempting to re-connect: 1/3
[14:46:14] MQTT user 'home' has configured password, connecting with credentials...
[14:46:14] Successfully established MQTT client connection!
[14:46:16] MQTT publishing failure: 'homeassistant/sensor/boiler_flow_temp_809c6a3c9c8/config': -9
[14:46:16] MQTT network disconnection detected trying to publish: 'homeassistant/sensor/boiler_flow_temp_809c6a3c9c8/config' attempting to re-connect: 1/3
[14:46:16] MQTT user 'home' has configured password, connecting with credentials...
[14:46:16] Successfully established MQTT client connection!
[14:46:19] MQTT publishing failure: 'homeassistant/sensor/dhw_flow_temp_target_809c6a3c9c8/config': -9
[14:46:19] MQTT network disconnection detected trying to publish: 'homeassistant/sensor/dhw_flow_temp_target_809c6a3c9c8/config' attempting to re-connect: 1/3
[14:46:19] MQTT user 'home' has configured password, connecting with credentials...
[14:46:19] Successfully established MQTT client connection!
[14:46:21] MQTT publishing failure: 'homeassistant/sensor/sh_flow_temp_target_809c6a3c9c8/config': -9
[14:46:21] MQTT network disconnection detected trying to publish: 'homeassistant/sensor/sh_flow_temp_target_809c6a3c9c8/config' attempting to re-connect: 1/3
[14:46:21] MQTT user 'home' has configured password, connecting with credentials...
[14:46:21] Successfully established MQTT client connection!
[14:46:22] MQTT publishing failure: 'homeassistant/sensor/sh_flow_temp_target_809c6a3c9c8/config': -9
[14:46:22] MQTT network disconnection detected trying to publish: 'homeassistant/sensor/sh_flow_temp_target_809c6a3c9c8/config' attempting to re-connect: 2/3
[14:46:22] MQTT user 'home' has configured password, connecting with credentials...
[14:46:22] Successfully established MQTT client connection!
[14:46:23] MQTT publishing failure: 'homeassistant/sensor/sh_flow_temp_target_809c6a3c9c8/config': -9
[14:46:23] MQTT network disconnection detected trying to publish: 'homeassistant/sensor/sh_flow_temp_target_809c6a3c9c8/config' attempting to re-connect: 3/3
[14:46:23] MQTT user 'home' has configured password, connecting with credentials...
[14:46:24] Successfully established MQTT client connection!
[14:46:24] Failed to publish homeassistant sh_flow_temp_target_809c6a3c9c8 entity auto-discover
[14:46:24] Software reset via esp_restart.
[14:46:24] Ecodan HomeAssistant Bridge startup successful, starting request processing.
[14:46:24] Watchdog initialized.
[14:46:24] Login cookie is unset, redirecting to login page
[14:46:28] Login cookie is unset, redirecting to login page
[14:46:30] Successful device login, authorising client.
```

limkinZero, 2024-05-01:
I think this issue can be closed. There are some diagnostic sensors.

closed 2024-05-03 by rbroker