rbroker/ecodan-ha-local #25: Z2 temp issue (SH instead)

thoniodu10, 2024-03-05:
Hi Team,

First, THANK YOU for that great work ! After some times to setup my config Ecodan + ESP32-S2 + HA over MQTT, everything is now going well.

There is only some stange things...
I have two wireless Mitsu thermostat PAR50 + receiver PAR51...

For Z1, 
- z1 room temp is reported correctly in the sensor field
- z1 temp control is not working but the value reported is correct in the sensor filed (z1_room_temp ok).

For Z2, 
- room temp is reported nowhere.
- temp control value is reported with sh_room_temp. What is sh temp ?

I know that the issue is probably due to my installation, but is you have some hints for helping me, would be great.
Maybe I need to change some connections or setup ?

[image]
[image]
[image]

Many Thanks for your help !
Kind Regards,
Anthony.

rbroker, 2024-03-05:
> temp control value is reported with sh_room_temp. What is sh temp ?

SH is supposed to stand for space heating, but actually as you said it looks like a misunderstanding from my side, and actually it's the target room temperature value for zone 2. I'll probably include it in a climate control for Z2 in a future update, same as the current climate control for Z1 shows the current/target temps.

> z1 temp control is not working

Could you confirm in what way? Is there anything in the log on the "Diagnostics" page on the ESP32 indicating why it might have failed?

Thanks,
Richard

mharizanov, 2024-03-05:
z1 temperature control works for me..

thoniodu10, 2024-03-05:
Thanks for you quick feedback.
sorry for Z1 control, my bad.
I was using the slide below :
[image]

but it's not the temp target for z1 I guess...

Using the round slide, it's working great !

[image]

For Z2, I confirm... there is strange thing...

Here, it's fine 👍 (the 17.5 for Z2)
[image]

And here it's reported in SH... and I don't know what is space heating.

[image]

Anyway, if you need someone to test Z2 climate, I'm available.

But thanks again for your work !!
Kind Regards,
Antho.

lunarok, 2024-03-09:
I confirm same for me :
- sh_flow_temp_target -> it's the z2 flow temp target (in my case, Z2 is configured with flow temp, not room temp)
- z2_room_temp -> this one I don't know what it could be, 28° is not matching anything (flow is 45, target is 19, room is around 20 I have'nt go upstairs to check on screen, but for sure not 28 with 12 outside)

[image]

rbroker, 2024-03-09:
Hi,

I think the best thing would be to try enabling "dump serial packets" in the configuration, then share a copy of the diagnostic logs + the actual current room temp from zone 2, hopefully something in the commands matches that temp.

lunarok, 2024-03-09:
[19:10:31] RES { .Hdr { fc, 62, 2, 7a, 10 } .Payload { b, 7, d0, 7, d0, f0, c4, b, 9, c4, 7e, 5e, 0, 0, 0, 0 } .Chk { f1 } }
b here it's the 0x0B you are looking for right ?

At time of this info, I can see on sensor display :
Z1 : 20°
Z2 : 20°
(both same)

Do I understand correctly that this :
status.Zone1RoomTemperature = res.get_float16(1);
status.Zone2RoomTemperature = res.get_float16(7);

Will have to be more
status.Zone1RoomTemperature = res.get_float16(1);
status.Zone2RoomTemperature = res.get_float16(3);

Not sure of the index, but can see 7, d0, 7, d0, like twice the same info

limkinZero, 2024-03-10:
Hi. Yes, I remember when I started with the Ecodan and the esp32 that I did not have the interior control temperature probe configured and the integration reported that crazy temperature.

lunarok, 2024-03-10:
Looking fine with the new code 1.0.7 for me
[image]

rbroker, 2024-03-11:
> Let's see if I have time, and we put a flag in the configuration to enable or not the sensors and control of zone 2. On machines that don't have this, it doesn't make sense to show this. Something like cooling mode.

Yeah, I agree it would be nice if we can hide the 2-zone data if we detect that the system is only in a 1-zone configuration. I'll raise another issue as a feature request for that and close this one off as the Z2 temp at least seems ok now.

Thanks all!

closed 2024-03-11 by rbroker