rbroker/ecodan-ha-local #27: Compile Error LOLIN S2 Mini

lafite64, 2024-03-08:
Hi

Thanks for all the work on this, really appreciated.

I'm getting a compile error with Arduino IDE: looks like something is being substituted, but not sure where to look: any thoughts?

I'm using the Arduino ESP32 Boards library, picked out the LOLIN S2 Mini

Regards

Rod

M:\OneDrive\Documents\Personal\Woodside\Ecodan\ecodan-ha-local-main\ecodan-ha-local\ehal_diagnostics.cpp: In function 'void ehal::init_watchdog()':
M:\OneDrive\Documents\Personal\Woodside\Ecodan\ecodan-ha-local-main\ecodan-ha-local\ehal_diagnostics.cpp:151:27: error: invalid conversion from 'int' to 'const esp_task_wdt_config_t*' [-fpermissive]
  151 |         esp_task_wdt_init(30, true); // Reset the board if the watchdog timer isn't reset every 30s.
      |                           ^~
      |                           |
      |                           int
M:\OneDrive\Documents\Personal\Woodside\Ecodan\ecodan-ha-local-main\ecodan-ha-local\ehal_diagnostics.cpp:151:26: error: too many arguments to function 'esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t*)'
  151 |         esp_task_wdt_init(30, true); // Reset the board if the watchdog timer isn't reset every 30s.
      |         ~~~~~~~~~~~~~~~~~^~~~~~~~~~
In file included from M:\OneDrive\Documents\Personal\Woodside\Ecodan\ecodan-ha-local-main\ecodan-ha-local\ehal_diagnostics.cpp:16:
C:\Users\roddy\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32s2/include/esp_system/include/esp_task_wdt.h:47:11: note: declared here
   47 | esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t *config);
      |           ^~~~~~~~~~~~~~~~~

exit status 1

Compilation error: invalid conversion from 'int' to 'const esp_task_wdt_config_t*' [-fpermissive]

rbroker, 2024-03-09:
Hi @lafite64,

I'm not seeing any compile failures for the S2 Mini with esp32 2.0.14 (latest) and Arduino IDE 2.2.1. Can you confirm which versions you're using?

Holdys, 2024-04-04:
Yes.In Arduino IDE 1.8.19 (for Windows 7) is this failure.

rbroker, 2024-04-04:
Hi @Holdys,

There's two options then I think:
- Update to use newer ESP32 board support package / Arduino IDE
- Modify this code on your side to match the older versions

Since this was opened I've added some GitHub actions, so I'm confident that with an updated environment it can build for S2 mini without problems.

Thanks,
Richard

closed 2024-04-04 by rbroker