rbroker/ecodan-ha-local #29: Is it possible to add ethernet to the project?

rudireith, 2024-03-10:
At first thanks to all who was included for this amazing work! Works really excellent.

The only thing witch is not perfekt is my wlan in the house. So I was thinking about to buy a esp32 Modul with ethernet on board (WT32-ETH01), but I´m not able to put the needed lines in the code.

Is it possible to add ethernet to the project? This would be really nice.

Many thanks. Best regards Rudi.

rbroker, 2024-03-10:
Hi @rudireith,

There's no technical reason it can't be supported, I think but it's a relatively big change as the current code assumes you're using WiFi and espressif keep the wifi / ethernet code very separate.

It's not very practical for me to work on as I can't easily test any of it with my system (it's very far from the nearest ethernet socket in my house).

rudireith, 2024-03-10:
It was worth a try, thank you very much.

closed 2024-03-10 by rudireith