rbroker/ecodan-ha-local #34: Random hanging and rebooting.

itrankolov, 2024-03-17:
Hello,

I am running S2 Mini board (but tried with WROOM-32 and it was the same) and from time to time the board goes offline and after a while it reboots. Here is the log from the WEB GUI:
[17:36:15] CMD { .Hdr { fc, 42, 2, 7a, 10 } .Payload { a2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 } .Chk { 90 } }
[17:36:15] RES { .Hdr { fc, 62, 2, 7a, 10 } .Payload { a2, 18, 3, 10, 0, 29, 3c, 0, 0, 0, 0, b, 8, 0, 0, 0 } .Chk { cd } }
[17:47:22] Configuration parameters loaded from NVS
[17:47:22] Initializing WiFi connection...
[17:47:22] Waiting 500ms for WiFi connection...
[17:47:23] Waiting 500ms for WiFi connection...
[17:47:23] WiFi connection established!
[17:47:23] Updated UTC time from NTP
[17:47:23] Regular startup mode, initializing web-server...
[17:47:23] Initializing HeatPump with serial rx: 34, tx: 33
[17:47:23] CMD { .Hdr { fc, 5a, 2, 7a, 2 } .Payload { ca, 1, 5d, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 } .Chk { 0 } }
[17:47:23] Initializing MQTT...
[17:47:23] MQTT user 'dex' has configured password, connecting with credentials...
[17:47:23] RES { .Hdr { fc, 7a, 2, 7a, 1 } .Payload { 0, 9, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 } .Chk { 0 } }
[17:47:23] connection reply received from heat pump
[17:47:24] Successfully established MQTT client connection!
[17:47:27] Software reset via esp_restart.
[17:47:27] Ecodan HomeAssistant Bridge startup successful, starting request processing.
[17:47:27] Watchdog initialized.

Please advice on how to proceed with troubleshooting.

rbroker, 2024-03-17:
Looking at the ~10 minute gap, my best guess is that the board is becoming disconnected from your WiFi, and because it's not becoming available again within 10 minutes, the board is restarting itself (the `reboot_if_wifi_disconnected_too_long` method in ehal-ha-local.ino).

itrankolov, 2024-03-17:
It seems so, but why ? Signal is ~-60. I see that it timeouts on the APs:
2024/03/17 21:57:26.919572117 {wncd_x_R0-0}{1}: [dot11] [15422]: (info): MAC: 8065.994e.1030  Sent deauth to client, deauth reason: 252, CLIENT_DEAUTH_REASON_ADMIN_RESET delete reason: 24, CO_CLIENT_DELETE_REASON_SESSION_TIMEOUT.
It timeouts approx every 30-40 minutes.

rbroker, 2024-03-17:
> It seems so, but why ?

According to some [cisco documentation](https://www.cisco.com/c/en/us/support/docs/wireless/catalyst-9800-series-wireless-controllers/213970-catalyst-9800-wireless-controllers-commo.html#toc-hId-1692432591) this kind of message from the AP: 

> Reason:
> 
> The client reached its session timeout.
> 
> Possible solutions:
> 
>    This is a normal behavior
>    Increase the policy profile session timeout associated with the SSID

So it seems to be related to your network configuration somehow.

> Also when it timeouts it does not attempt to reconnect until it resets iself.

I'm using the reconnection feature built into the [ESP32 WiFi API](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/wifi.html#setautoreconnect), it seems to work well in my environment (e.g. reconnecting after the router is rebooted), I guess the same WiFi configuration you have which is causing your session timeout may also affect the auto-reconnect.

itrankolov, 2024-03-17:
This is strange, because I have 10 more ESPHome devices connected to the same network, and they don't have issues. Can I somehow debug this over the USB port ?

rbroker, 2024-03-19:
> This is strange, because I have 10 more ESPHome devices connected to the same network, and they don't have issues

Looking at the esphome code, they've implemented their own WiFi connection logic, including their own reconnect mechanism (they have a huge amount of code to handle it)

>  Can I somehow debug this over the USB port ?

There's not much from the point of view of this project which you can log over the serial monitor. The WiFi is just set up, and I toggle the auto-reconnect on, and leave it.

You could try enabling auto-reconnect in one of the [Arduino ESP32 samples](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientConnect/WiFiClientConnect.ino) and seeing if it has the same problem of eventually disconnecting, and if so raise an issue with them - they may have more knowledge of what WiFi diagnostic information they can collect.

Alternatively, as you already have lots of esphome devices, you could try something like: https://github.com/tobias-93/esphome-ecodan-heatpump as that should give you access to their WiFi handling logic which seems to work better than the espressif one in your environment.

itrankolov, 2024-03-19:
Thanks for your reply. I have changed the reboot_if_wifi_disconnected_too_long interval to 1 minute and modified the WiFi setup  litttle bit, it still disconnects but not that often.

jamessimpson007, 2024-05-01:
Thanks. I too am having problems with WiFi signal and connection. The WiFi signal sensor has been very useful to help diagnose, but has demonstrated that my signal is very variable, due to the ESP32 (Wroom) board often choosing to inappropriately connect to a far away AP with a low signal. I have an orbi WiFi system with a router and 5 peripheral satellite APs, all running the same SSID. Although I get a choice of all the different APs on the configuration page, as they all have the same user/pass the ESP32 board freely switches between them, seemingly at random. When connected to the router (a couple of metres away) I get a perfect WiFi signal of -50dB. However it doesn't last long before it switches to a different AP with a much worse signal. 

I have tried setting "maxWifiDisconnectLength" in ecodan-ha-local.ino to 2s rather than 10s, which has helped a bit as sometimes after a reboot it connects to a better AP temporarily, before switching again back to a further away one again. Occasionally it looses the connection altogether but eventually reboots and broadcasts its own AP again for me to re-configure. 

From browsing the internet it seems that this is a fairly common problem with ESP32 boards. I'm toying with the idea of trying to home-brew an external antenna by scratching off the internal WiFi aerial connections. Has anyone else found a similar problem / solution?

Would it be easy to implement something along these lines, to only choose the WiFi access point with the strongest signal? https://www.esp32.com/viewtopic.php?f=19&t=18979&start=10

Thanks all for your help, it really is an excellent project!

rbroker, 2024-05-03:
> Would it be easy to implement something along these lines, to only choose the WiFi access point with the strongest signal? https://www.esp32.com/viewtopic.php?f=19&t=18979&start=10

That's interesting. Yes, I think it should be possible to add a setting which does something like:
- When scanning, save a copy of the BSSID which has the highest signal strength
- When calling WiFi.begin() passing the saved BSSID to that method.

jamessimpson007, 2024-06-26:
In the mean time, I've creating a separate WiFi network on my LAN specifically for the esp32 board using an old router I had lying around. As it only has one AP using the SSID it no longer jumps around between APs. It was getting to the stage where it was frequently locked onto distant APs within my mesh network, with very weak WiFi signals and therefore unstable connections to the MQTT broker. As you can see it is now a much more stable connection and working well. 

[image]

jamessimpson007, 2024-09-30:
In case anyone is interested, despite the above helping I still had issues with my module (ESP32 Wroom) intermittently becoming unavailable on my network and unresponsive to ping, and not connecting to MQTT. I think it's something to do with DHCP / address reservation but despite extensive experimenting I've been unable to resolve it. I have its IP address reserved on my DHCP server. It is usually resolved with a manual restart (shorting pins), so I've implemented a pretty crude and inelegant solution of rigging up a spare zigbee relay to automatically short the reset pins using a HA based automation if the MQTT connection is disconnected for more than a few minutes. Has been working well for a few months with a reset once or twice a week and will do for now....