rbroker/ecodan-ha-local #36: Any thoughs on using ESPHome?

gekkekoe, 2024-03-19:
Recently I migrated to ESPHome variant of this project: https://github.com/vekexasia/comfoair-esp32
It was really easy, and most of the boiler plate code is gone (all those pesky dependencies/build envs).

configuration (pins, sensors) is done via yaml for example: https://github.com/vekexasia/comfoair-esp32/blob/master/docs/esphome_sample.yaml
It also has a web interface where it shows all sensors and msg in realtime. it also has basic controls (on/off switches, set values)
for example: https://esphome.io/_images/web_server-v2.png

Building is as easy as:
`esphome compile file.yaml`
no need for build envs. Just python will do, or use the docker images.

OTA:
`esphome upload  --device UPLOAD_PORT  (/dev/ttyUSB0, or ip address)`

you can also manage it all via home assisant https://esphome.io/guides/getting_started_hassio.html

There are other projects that takes the ESPHome approach for the mitsubishi heatpumps but in my opinion this codebase is better.
Any thoughs on using ESPHome as base for this project?

rbroker, 2024-03-19:
I don't currently have any plan to switch over to using esphome as a base, although I was looking at it the other day for issue #34. It's a fairly big change for my setup at the moment (it would actually be quite a lot of work to unpick this project from all the boilerplate code from the espressif APIs and move to esphome style components), and as you noted other people are already working on esphome components to fill the same purpose.

gekkekoe, 2024-04-12:
Since I wanted to learn how to make an esphome component, I went ahead ad created an esphome version of ecodan ha local. 
Been using it for a week now, and quite happy with the results. see https://github.com/gekkekoe/esphome-ecodan-hp

rbroker, 2024-04-13:
Very nice!

closed 2024-06-29 by rbroker