rbroker/ecodan-ha-local #38: Always switch back to "Heat Target Temperature"

dinkel1975, 2024-03-21:
Hi,
I don't get it to work. 
No matter which heating mode I set, after a short time it jumps back to "Heat Target Temperature". Can you help me?
Best regards
Michael

rbroker, 2024-03-30:
Hi,

Usually switching back means the command home assistant tried to send hasn't worked correctly, so the state is set back the next time the sensor status is updated.

Is it possible to see if there's anything suspicious in the log messages on the diagnostics tab of the ESP32 web UI when you try and switch the mode?

Thanks,
Richard

dinkel1975, 2024-03-30:
hi,

In the meantime I have tried the Image from here https://github.com/tobias-93/esphome-ecodan-heatpump
With this Firmware mode switching is OK. 

Ciao 
Michael

rbroker, 2024-03-30:
Ok! No problem

closed 2024-03-30 by rbroker