rbroker/ecodan-ha-local #40: Auto-Discovery in HomeAssistant Not Working

merlin051, 2024-04-08:
I'm runing this sketch on a ESP32 WROOM-32D, flashed the sketch and configured the wifi to jump on my network, no issues there.

I'm running the latest HA on a rpi3-64 I'm not seeing any auto-discovery happening, the ESP is also not yet connected to the EcoDan as its just on bench right now.

I purposley have not configured MQTT on the ESP - is this needed for auto discovery?

[image]

rbroker, 2024-04-09:
Hi,

> I purposley have not configured MQTT on the ESP - is this needed for auto discovery?

Yes, the auto-discovery works using MQTT, so it needs to be configured before the ESP32 is able to communicate with homeassistant (and if you're not using the MQTT broker add-on for homeassistant you'll have to also set up homeassistant to access the broker.

Similarly, it won't send any information over MQTT if it's not connected to the heat pump, as there's essentially no valid information it can provide in that state.

Thanks,
Richard

closed 2024-04-09 by rbroker

merlin051, 2024-04-09:
I think the thing that threw me here is on your images it doesn't show as MQTT being the integration broker so I wondered if it was using some other discovery mechanism.

I've installed the mosquito mqtt broker and configured HomeAssistant MQTT. The ecodan was discovered after and I now have entities listed in HomeAssistant.(Although no valid data reported whilst it's still on the bench)

I'll be doing some testing with my 11.2kw ecodan over the coming days/weeks.

This is a fantastic repo, thanks for the effort!!

merlin051, 2024-04-09:
for reference, mine looks like this whilst on the bench:
[image]

rbroker, 2024-04-09:
> I think the thing that threw me here is on your images it doesn't show as MQTT being the integration broker so I wondered if it was using some other discovery mechanism.

Ah sorry, I'll look at getting the images updated at some point, they're quite old at this point and try to clarify the requirements in the readme a bit.

> I'll be doing some testing with my 11.2kw ecodan over the coming days/weeks.

Ok, good luck! Hopefully everything goes smoothly!