rbroker/ecodan-ha-local #43: Compile error LOLIN S2 Mini

bikerbas74, 2024-06-18:
Hi I get a Compile errors, I do not know how to get rid of them.
Help is appreciated.

I use:
Arduino ide V 2.3.2

Board:
ESP32 Espressif Systems V 2.0.17

Library:
Arduinojson Benoit Blanchon V 7.0.3
Seeed_Arduino_mbedtls Peter Yang V 3.0.1
MQTT Joel Gaehwiler V 2.5.2

I'm not sure if these libs are correct, the description is of the dependencies is not very detailed.

Compile errors:

F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp: In function 'String ehal::http::generate_login_cookie()':
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:65:9: error: 'mbedtls_md_context_t' was not declared in this scope
         mbedtls_md_context_t ctx;
         ^~~~~~~~~~~~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:65:9: note: suggested alternative: 'md5_context_t'
         mbedtls_md_context_t ctx;
         ^~~~~~~~~~~~~~~~~~~~
         md5_context_t
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:67:26: error: 'ctx' was not declared in this scope
         mbedtls_md_init(&ctx);
                          ^~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:67:9: error: 'mbedtls_md_init' was not declared in this scope
         mbedtls_md_init(&ctx);
         ^~~~~~~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:67:9: note: suggested alternative: 'esp_rom_md5_init'
         mbedtls_md_init(&ctx);
         ^~~~~~~~~~~~~~~
         esp_rom_md5_init
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:68:58: error: 'MBEDTLS_MD_SHA256' was not declared in this scope
         mbedtls_md_setup(&ctx, mbedtls_md_info_from_type(MBEDTLS_MD_SHA256), 0);
                                                          ^~~~~~~~~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:68:32: error: 'mbedtls_md_info_from_type' was not declared in this scope
         mbedtls_md_setup(&ctx, mbedtls_md_info_from_type(MBEDTLS_MD_SHA256), 0);
                                ^~~~~~~~~~~~~~~~~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:68:9: error: 'mbedtls_md_setup' was not declared in this scope
         mbedtls_md_setup(&ctx, mbedtls_md_info_from_type(MBEDTLS_MD_SHA256), 0);
         ^~~~~~~~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:69:9: error: 'mbedtls_md_starts' was not declared in this scope
         mbedtls_md_starts(&ctx);
         ^~~~~~~~~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:70:9: error: 'mbedtls_md_update' was not declared in this scope
         mbedtls_md_update(&ctx, (const unsigned char*)payload.c_str(), payload.length());
         ^~~~~~~~~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:70:9: note: suggested alternative: 'esp_rom_md5_update'
         mbedtls_md_update(&ctx, (const unsigned char*)payload.c_str(), payload.length());
         ^~~~~~~~~~~~~~~~~
         esp_rom_md5_update
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:71:9: error: 'mbedtls_md_finish' was not declared in this scope
         mbedtls_md_finish(&ctx, sha256);
         ^~~~~~~~~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_http.cpp:72:9: error: 'mbedtls_md_free' was not declared in this scope
         mbedtls_md_free(&ctx);
         ^~~~~~~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_mqtt.cpp:29:5: error: 'MQTTClient' does not name a type; did you mean 'espClient'?
     MQTTClient mqttClient(4096);
     ^~~~~~~~~~
     espClient
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_mqtt.cpp: In function 'String ehal::mqtt::get_connection_error_string()':
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_mqtt.cpp:353:17: error: 'mqttClient' was not declared in this scope
         switch (mqttClient.lastError())
                 ^~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_mqtt.cpp:353:17: note: suggested alternative: 'espClient'
[... 109 lines cut ...]
                      espClient
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_mqtt.cpp:1342:9: error: 'mqttClient' was not declared in this scope
         mqttClient.loop();
         ^~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_mqtt.cpp:1342:9: note: suggested alternative: 'espClient'
         mqttClient.loop();
         ^~~~~~~~~~
         espClient
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_mqtt.cpp: In function 'bool ehal::mqtt::is_connected()':
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_mqtt.cpp:1347:16: error: 'mqttClient' was not declared in this scope
         return mqttClient.connected();
                ^~~~~~~~~~
F:\Storage\docs\Warmtepomp\eigenControl\warmtepomp\software\ecodan-ha-local-main\ecodan-ha-local-main\ecodan-ha-local\ehal_mqtt.cpp:1347:16: note: suggested alternative: 'espClient'
         return mqttClient.connected();
                ^~~~~~~~~~
                espClient
Multiple libraries were found for "Preferences.h"
  Used: B:\Users\biker\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17\libraries\Preferences
  Not used: B:\Users\biker\Documents\Arduino\libraries\Preferences
Multiple libraries were found for "WiFi.h"
  Used: B:\Users\biker\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17\libraries\WiFi
  Not used: B:\Users\biker\Documents\Arduino\libraries\WiFi
exit status 1

Compilation error: 'mbedtls_md_context_t' was not declared in this scope

rbroker, 2024-06-20:
Hi @bikerbas74,

> Seeed_Arduino_mbedtls Peter Yang V 3.0.1
> MQTT Joel Gaehwiler V 2.5.2
> 
> I'm not sure if these libs are correct, the description of the dependencies are not very detailed

The dependency versions and the authors you listed look correct, but the compile errors you posted indicate that the Arduino IDE isn't able to find these library files installed on your system, so I'd double check they're listed as installed by the library managed.

I don't see any compile errors here for the JSON library, so that one looks OK, it's just the mbedtls and MQTT libraries that seem to have an issue.

Thanks,
Richard

bikerbas74, 2024-06-20:
Hi Richard,

Thank you for your quick reply!

Good to know that the libraries are the correct ones.

I will investigate what's going on next week and let you know. 

Regards,
Bas

bikerbas74, 2024-06-22:
Hi Richard,

I concentrated first on Seeed_Arduino_mbedtls Peter Yang V 3.0.1.
Yes it was installed in lib manager and yes it was in my arduino lib folder.

I opened the example file delivered by the library HelloHMACSHA256.ino, compiled it and it did not work.
I downgraded to version 2.01 loaded the example file delivered with that version "nothing" which actually did nothing, just include the lib and not using it (it is a good name for the file ;-)), also there compile errors.

Manually downloaded 3.0.1 from github, installed manual, try to compile the example => same story
I downloaded version from DEV (https://github.com/Seeed-Studio/Seeed_Arduino_mbedtls/tree/dev) installed it manually, -> also no luck, same compile errors.

replaced #include "Seeed_mbedtls.h" with #include <Seeed_mbedtls.h> did not work as well.

Github page from the lib did not leave me a clue either.
I do not get this thing to work...about to give up.

rbroker, 2024-06-28:
Hi @bikerbas74,

Unfortunately, it sounds like there's some problem with your arduino IDE installation, if you've installed the libraries through the library manager, it should "just work". The only thing I could suggest is to completely reinstall the arduino IDE from scratch as it sounds like it's in some strange state...

> What version of the board lib are you using?
> I tried esp32 Espressif Systems 2.0.17 and 3.0.1.

I was using 2.0.17 to compile before, 3.x.x does not currently work correctly as they changed some function definitions in the new version.

closed 2024-06-28 by rbroker

bikerbas74, 2024-07-05:
Hi Richard,

 

Just FYI:

 

I got the previous code to compile using explicit path to lib:

#include "B:\Users\biker\Documents\Arduino\libraries\Seeed_Arduino_mbedtls\src\Seeed_mbedtls.h"

 

Your previous version before #47 compiles and uploads.

Connection to ecodan wifi network is possible and config page is visible.

You can select the desired WIFI  from a SSID WIFI network list.

Unfortunatelly  once selected and password set correct, it does not connect.

After 20 min automatic returns to ecodan published network and flashing blue led.

 

 

I tried new version including #47 with esp 3.0.2 board lib and no other wifi lib.

This compiles and uploads, but the list with wifi networks is empty…

 

No luck with this one ;-)

 

Regards,

Bas