rbroker/ecodan-ha-local #44: Compile Error

fredcazaux, 2024-06-26:
Hello, I'm just trying to compile, but face issues : 

`E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_diagnostics.cpp: In function 'void ehal::init_watchdog()':
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_diagnostics.cpp:151:27: error: invalid conversion from 'int' to 'const esp_task_wdt_config_t*' [-fpermissive]
  151 |         esp_task_wdt_init(30, true); // Reset the board if the watchdog timer isn't reset every 30s.
      |                           ^~
      |                           |
      |                           int
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_diagnostics.cpp:151:26: error: too many arguments to function 'esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t*)'
  151 |         esp_task_wdt_init(30, true); // Reset the board if the watchdog timer isn't reset every 30s.
      |         ~~~~~~~~~~~~~~~~~^~~~~~~~~~
In file included from E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_diagnostics.cpp:16:
C:\Users\Fred-GL503\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32s2/include/esp_system/include/esp_task_wdt.h:47:11: note: declared here
   47 | esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t *config);
      |           ^~~~~~~~~~~~~~~~~
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_http.cpp: In function 'String ehal::http::configuration_status()':
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_http.cpp:54:32: error: 'WiFi' was not declared in this scope
   54 |         if (config.HostName != WiFi.getHostname())
      |                                ^~~~
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_http.cpp: In function 'void ehal::http::handle_query_ssid_list()':
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_http.cpp:270:26: error: 'WiFi' was not declared in this scope
  270 |         int16_t result = WiFi.scanComplete();
      |                          ^~~~
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_http.cpp:272:23: error: 'WIFI_SCAN_RUNNING' was not declared in this scope
  272 |         if (result == WIFI_SCAN_RUNNING)
      |                       ^~~~~~~~~~~~~~~~~
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_http.cpp:277:28: error: 'WIFI_SCAN_FAILED' was not declared in this scope
  277 |         else if (result == WIFI_SCAN_FAILED)
      |                            ^~~~~~~~~~~~~~~~
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_http.cpp: In function 'void ehal::http::handle_diagnostics()':
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_http.cpp:345:46: error: 'WiFi' was not declared in this scope
  345 |         page.replace(F("{{wifi_hostname}}"), WiFi.getHostname());
      |                                              ^~~~
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_http.cpp: In function 'bool ehal::http::initialize_captive_portal()':
E:\Documents\Fred\2024\Thermo\ecodan-ha-local\ehal_http.cpp:577:50: error: 'WiFi' was not declared in this scope
  577 |         if (!dnsServer->start(/*port =*/53, "*", WiFi.softAPIP()))
      |                                                  ^~~~

exit status 1

Compilation error: invalid conversion from 'int' to 'const esp_task_wdt_config_t*' [-fpermissive]`

Someone could give me any clue ?
Thanks a lot !

rbroker, 2024-06-26:
Hi @fredcazaux,

It looks like espressif made some breaking changes to the ESP32 APIs in version 3.x.x of the ESP32 board package. I get similar issues if I update my board package to that version. Downgrading back to 2.0.17 again seems to allow things to compile again.

I'll have a look at what needs changing to support building with the latest version.

fredcazaux, 2024-06-26:
Yes was about to test that... Actually did it, and it works ! Thanks !
Maybe you could mention to use this version the same way you tell the libraries versions to use... ? (the time you find a correction for this...)
Thanks again for your very fast answer !

closed 2024-06-26 by fredcazaux