rbroker/ecodan-ha-local #45: ESP remains in an undefined state

Himbeerlinux, 2024-06-28:
Hi
I have installed the software on an ESP32 and it works without MQTT, as described.
I can access the ESP via WEB. As soon as I activate the MQTT broker, the software stops somewhere.

I use the IObroker to collect all the data. There is a separate module for MQTT.
After the ESP starts, the connection to the MQTT broker is also briefly established and an object tree is partially created and data is stored. After about 15 seconds, this connection is disconnected and the ESP remains in an undefined state. Access via WEB does not work. I cannot read the log file. As soon as the broker is running, WEB access does not work at all. You can only unplug the ESP and start again. At some point the program freezes and stops. A ping to the IP still works, so the network connection is still active.
What can you do to find the error.

rbroker, 2024-06-28:
I'm not familiar with ioBroker, so it's difficult for me to say what might be happening in this case. I had a quick look over the ioBroker source code, and it looks like they're dispatching state data in response to the MQTT subscription requests, my best guess is that it's sending something in a format homeassistant wouldn't send (as I guess it doesn't use homeassistant's value template format) and it's not being handled properly.

You could try temporarily commenting out the code inside the `mqtt_callback()` function in `ehal_mqtt.cpp`, if you're only looking to collect back the reported data and no control the pump - if that stops the hanging issue then at least you've identified that it's something to do with the data being sent by ioBroker whhich my code can't currently handle.

Himbeerlinux, 2024-06-29:
Thanks for the quick reply. I don't know the homeassistant, but I will test it. It may replace the iobroker.
I will also test the changes in the code to see if it is sufficient for my purposes.

Himbeerlinux, 2024-06-30:
I changed the code. Now iobroker can handle the data and I can add the info to my Grafana dashboard. Thanks again.

closed 2024-07-31 by rbroker