rbroker/ecodan-ha-local #48: Problème MQTT

Guitton50, 2024-07-22:
Bonjour,
Je suis novice sur Home Assistant avec le MQTT et j'ai un problème pour connecter un ESP32 flashé avec les fichiers ecodan-ha-local.
Tout se passe bien lors du téléversement avec IDE Arduino mais lors de la configuration web, j'obtiens toujours un erreur MQTT 
[image]

LWMQTT_NETWORK_FAILED_CONNECT
Après de nombreuses recherches par moi même sur internet je m'en remets à vous.
Les paramètres MQTT sont bons car vérifiés en ligne et sur d'autres intégrations HA.
Pouvez-vous m'aider svp?
Merci d'avance

rbroker, 2024-07-31:
Je suis desolée, mais je ne parle pas Francais depuis que j'ai etait dans l'ecole...

Peut-tu montre moi exactamente que c'est que tu a mettre pour "MQTT Server" dans la configuration? Si tu utiliser le "Mosquitto broker" addon dans Home Assistant, ça devrait être seuelemnt le "IP Address" ou "Host" de Home Assistant (`192.168.1.180` ou `homeasisstant.local` par example, mais ça depend de la configuration de votre routeur).

Guitton50, 2024-07-31:
Merci de ton retour, j'ai mis l'IP de home assistant.
J'ai testé avec mqtt explorer sur un pc sous windows et mes paramètres
fonctionnent correctement .
Je ne comprends pas ce qui ne va pas 🤔🤔

rbroker, 2024-07-31:
> J'ai testé avec mqtt explorer sur un pc sous windows et mes paramètres
> fonctionnent correctement .

Hmm... et le "Port" est "1883"? Peut-être le mot de passe a les lettres Françaises, comme "ê", et il y a une probleme avec lui dans le code? Est-ce qu'il y a quelquechose dans le "log" de Mosquitto en home assistant autour de ses connections (`<homeassistant IP>:8123/hassio/addon/core_mosquitto/logs`)?

Guitton50, 2024-07-31:
Oui port 1883.
Mon mot de passe n'a pas d'accent et je ne vois rien dans les logs de
mosquitto.
J'ai tout essayé avant d'écrire ce poste 😁

khakhux, 2024-08-07:
Hi

Putting back line 
```mqttClient.begin(config.MqttServer.c_str(), config.MqttPort, espClient);``` 
in _ecodan-ha-local/ehal_mqtt.cpp_,  fixed this for me.

This was deleted on commit 55aea33b881a74b3dcf5d3837246ec7a76ea6e18 for  #47

khakhux, 2024-08-27:
Hi. 

Something similar happened to me yesterday. 

Mosquitto logs didn't show any connection attempt.

I modified the code to retry a few times the connection to MQTT server and it worked for me.

I have added PR #50 with a proposed solution.

Cheers!

Carlos

commit f999c14a (2024-08-28): Add mqtt connection retries (#50)

* fix: #48 Add MQTT server connection retries

* fix: #48 update version to 0.2.2
closed 2024-08-28 by rbroker