rbroker/ecodan-ha-local #49: possibility to activate or deactivate heating/cooling

quentindavid, 2024-07-30:
Hi,
Would it be possible to add the control of heating/cooling ? At the moment we can only turn on of off completely the unit ?

Also is it normal that I can't set temperature for flow in cooling mode ? It reports the correct temperature but if I change it it send a command to set the new temperature, and soon after that it come back to its original value.

rbroker, 2024-07-31:
Hi,

> Would it be possible to add the control of heating/cooling ? At the moment we can only turn on of off completely the unit ?

Can you clarify what you mean by this? It should already be possible to set the target temperatures and change the modes. Is there extra control in your system which is currently missing?

> Also is it normal that I can't set temperature for flow in cooling mode ? It reports the correct temperature but if I change it it send a command to set the new temperature, and soon after that it come back to its original value.

That means your controller isn't accepting the command to change the temperature correctly. I don't have cooling support on mine, only heating - so it's possible your controller works differently to the one owned by the person who origianlly added cooling support.

quentindavid, 2024-07-31:
Yes I mean, I can see I can change the target temperature and change the mode, but on the unit I can choose to enable to disable cooling/heating. It displays a logo "play", "scheduled" or "disabled". It would be nice to control the same remotely (I'd like to couple this to my solar panels in order to pause cooling when not enough power available).

My controller looks very close to yours as I have an Ecodan ERST20D-VM6DR1 and external unit PUHZ-SW75VAA. I can see it's a FTDC6 unit also (I have the manual with this reference on top)

Maybe I can help in order to send different commands to see which one would work ? Is it possible to send different tries ?

rbroker, 2024-08-01:
> It displays a logo "play", "scheduled" or "disabled". It would be nice to control the same remotely (I'd like to couple this to my solar panels in order to pause cooling when not enough power available).

It looks like [this project](https://github.com/gekkekoe/esphome-ecodan-hp) has found the "prohibit" flag for heating/cooling and has an experimental implementation already (I think it's equivalent to "disabled" you mentioned). It's possible the flag to  change to scheduled modes is similar.

> Maybe I can help in order to send different commands to see which one would work ? Is it possible to send different tries ?

I think that's a bit risky, as searching for new settings commands this way may put your system into a state which is difficult or impossible to undo.

quentindavid, 2024-08-01:
hi @rbroker, thanks for the info I'll check this out and let you know if it's work !

quentindavid, 2024-08-02:
So I tested and it turns out it works quite well in my case ! It can control the setpoint temperature for water when cooling is enabled. It can also activate "server mode", which override the parameters on the unit itself. Then heating/cooling/dhw is controlled from HA only (a new icon of earth appears on the FTC6 display instead of prohibit sign).
It could be nice in order to drive DHW production and heating/cooling as well depending on the PV production in real time.

JeromeCarsonDomin, 2025-07-22:
Awesome job @rbroker! This works a treat, it took me a little while to get set up on an ESP32 S2 Mini (just needed to add a resistor on one of Tx/Rx wires and all good! I do have one missing piece: the operation modes don't show cooling. When I select cooling on the unit itself the state on HA shows 'unknown' (see background log in the picture), and cooling doesn't show on the option list? Does anyone have any ideas?

[image]

JeromeCarsonDomin, 2025-07-22:
> Awesome job [@rbroker](https://github.com/rbroker)! This works a treat, it took me a little while to get set up on an ESP32 S2 Mini (just needed to add a resistor on one of Tx/Rx wires and all good! I do have one missing piece: the operation modes don't show cooling. When I select cooling on the unit itself the state on HA shows 'unknown' (see background log in the picture), and cooling doesn't show on the option list? Does anyone have any ideas?
> 
> <img alt="Image" width="593" height="622" src="https://private-user-images.githubusercontent.com/70695892/469451370-1984f9df-6da1-4f49-9c6a-65e4df0bac18.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTMyMTg4NzEsIm5iZiI6MTc1MzIxODU3MSwicGF0aCI6Ii83MDY5NTg5Mi80Njk0NTEzNzAtMTk4NGY5ZGYtNmRhMS00ZjQ5LTljNmEtNjVlNGRmMGJhYzE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzIyVDIxMDkzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZlZTAzNDZjZjY2YjdjM2FiNDQ2YmE1NGRhMjA2ZWU2MjUzMTYyYThiMDJkOTNiYTJhZWVlYWI5YmRkNmMzMGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.6Bl-FV9qB19sC5KpLebSK7f1D-MeT8pr84ikVvG9ytA">

Apologies!! I completely missed the enable cooling on the configuration webpage! All working! <3